dmamontov/esphome-blekeyboard #1: Whenever the ESP reboots/resets, I have to re-pair it to my device

jonathanrobichaud4, 2021-11-21:
First I want to say thank you so much for making this a real thing! I've been searching for a solution like this for a while! 

The title says most of it. Whenever the ESP looses power I have to repair it to my Google TV. However, it doesn't loose connection when I reboot my Google TV. Just for clarity it's running android 10.

Is this something I should be asking in the ESP32-BLE-Keyboard library?

Also, I'm working on a custom card that can be used to input text which reflects the text box on the tv (for the most part). I can link the repository once I create it if that is of interest?

Thanks again!

dmamontov, 2021-11-21:
> Is this something I should be asking in the ESP32-BLE-Keyboard library?

Most likely it's really about the libraries. But I'll check(Android TV) it in the week, it's interesting myself) They also have a beta, so I'm thinking of switching to it, she can do more. We also plan to make a usb hid keyboard.

> Also, I'm working on a custom card that can be used to input text which reflects the text box on the tv (for the most part). I can link the repository once I create it if that is of interest?

Oh sure. It would be helpful.

jonathanrobichaud4, 2021-11-21:
Oh ok! I just switched to the beta library and that fixed my problem! Here is the link for the card: https://github.com/jonathanrobichaud4/lovelace-live-text-update-card. I'll make my first commit soon but, I have to clean it up a bit.

closed 2021-11-21 by jonathanrobichaud4

dmamontov, 2021-11-22:
> Oh ok! I just switched to the beta library and that fixed my problem! Here is the link for the card: https://github.com/jonathanrobichaud4/lovelace-live-text-update-card. I'll make my first commit soon but, I have to clean it up a bit.

Added a new version, in addition to updating the library, there are still many changes)

https://github.com/dmamontov/esphome-blekeyboard/releases/tag/v1.0.1

jonathanrobichaud4, 2021-11-22:
I can open a new issue but when I try to compile the new firmware I get this error:
`/config/esphome/remote-and-temp-sensor.yaml: In lambda function:
/config/esphome/remote-and-temp-sensor.yaml:53:56: error: 'class Esp32BLEKeyboard' has no member named 'send_message'
           (static_cast<Esp32BLEKeyboard *>(ble_keyboard))->send_message(message, is_combination, delay_ms);`

I changed the name to remote-and-temp-sensor and as you can probably tell I use a temp sensor with it. Do you know what is going on?

jonathanrobichaud4, 2021-11-22:
I just saw that you made a commit and that is working!

dmamontov, 2021-11-22:
> I just saw that you made a commit and that is working!

everything was very trite) I didn't push everything) take a fresh master)