dmamontov/esphome-blekeyboard #3: Doesn't compile on latest ESPHome

ThisIsTheOnlyUsernameAvailable, 2022-04-25:
Hi!

Thanks for this library.  It's a fantastic idea.

It doesn't look like it compiles on the latest ESPHome.  I've tried with both Arduino and Espressif frameworks without success.

My basic YAML:

```
esphome:
  name: bt-test
  platformio_options: # Delete if you don't want to use NimBLE mode. More details: https://github.com/T-vK/ESP32-BLE-Keyboard#nimble-mode
    build_flags: -D USE_NIMBLE
  includes:
    - libraries/ble-2/keyboard.h
    - libraries/ble-2/keymap.h

  libraries:
    - ESP32 BLE Arduino@>=1.0.1
    - https://github.com/h2zero/NimBLE-Arduino/archive/refs/tags/1.3.2.zip
    - https://github.com/T-vK/ESP32-BLE-Keyboard/releases/download/0.3.1-beta/ESP32-BLE-Keyboard.zip

  
esp32:
  board: tinypico
  framework:
    type: arduino
    version: recommended

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "<removed>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bt-Test Fallback Hotspot"
    password: "<removed>"

# captive_portal:
```

Compiling under Arduino does the following:

```
INFO Reading configuration /config/bt-test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing bt-test (board: tinypico; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing https://github.com/h2zero/NimBLE-Arduino/archive/refs/tags/1.3.2.zip

Unpacking  [####################################]  100%
Library Manager: NimBLE-Arduino @ 1.3.1 has been installed!
Library Manager: Installing https://github.com/T-vK/ESP32-BLE-Keyboard/releases/download/0.3.1-beta/ESP32-BLE-Keyboard.zip

Unpacking  [####################################]  100%
Library Manager: ESP32 BLE Keyboard @ 0.3.1 has been installed!
Dependency Graph
|-- <ESP32 BLE Arduino> 1.0.1
[... 152 lines cut ...]
Compiling .pioenvs/bt-test/src/main.o
Generating LD script .pioenvs/bt-test/esp32_out.ld
In file included from src/main.cpp:15:
src/keyboard.h:4:10: fatal error: BleKeyboard.h: No such file or directory

*********************************************************************
* Looking for BleKeyboard.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:BleKeyboard.h"
* Web  > https://registry.platformio.org/search?q=header:BleKeyboard.h
*
*********************************************************************

 #include <BleKeyboard.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/bt-test/src/main.o] Error 1
========================= [FAILED] Took 10.01 seconds =========================
```

Trying to meet the dependency just goes around in circles.

Any suggestions?

TIA.

jonathanrobichaud4, 2022-04-25:
Try using the latest version of nimble and blekeyboard (the links are in the readme) I was having issues with my esp constantly crashing and using the latest versions fixed that so maybe it will help.

ThisIsTheOnlyUsernameAvailable, 2022-04-25:
> Try using the latest version of nimble and blekeyboard (the links are in the readme) I was having issues with my esp constantly crashing and using the latest versions fixed that so maybe it will help.

Thanks for the suggestion, but the issue remains with the latest releases of both libraries.

dmamontov, 2022-04-30:
@ThisIsTheOnlyUsernameAvailable I tried on version 2022.4.0 everything compiles correctly. Maybe your esphome itself has not been updated correctly?
[build.log](https://github.com/dmamontov/esphome-blekeyboard/files/8597214/build.log)

ThisIsTheOnlyUsernameAvailable, 2022-04-30:
I'm really sorry to have wasted your time, @dmamontov.  I'll try to work out why my setup doesn't work and will post it here, in case anyone else has the same issue.

dmamontov, 2022-07-31:
Not relevant. A new version has been released.

closed 2022-07-31 by dmamontov