dmamontov/esphome-blekeyboard #17: Compilation error

LucaLa2001, 2022-12-17:
When I try to install the firmware on the node, or even just download the binary, I get an error. Looking at the log I suspect it is caused by ble keyboard, as a matter of fact if I remove this component there are no errors.

I run Esphome as a Home Assistant plugin
Esphome plugin version: 2022.12.1
Home Assistant OS version: 8.5

**Log**:
```
INFO Reading configuration /config/esphome/cinema.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing cinema (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ESP32 BLE Arduino @ 1.0.1
|-- NimBLE-Arduino @ 1.4.0
|-- ESP32 BLE Keyboard @ 0.3.2
Compiling /data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLE2902.cpp.o
Compiling /data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLE2904.cpp.o
Compiling /data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLEAdvertisedDevice.cpp.o
Compiling /data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLEAdvertising.cpp.o
Compiling /data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLECharacteristic.cpp.o
Compiling /data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLECharacteristicMap.cpp.o
Compiling /data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLEClient.cpp.o
Compiling /data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLEDescriptor.cpp.o
Compiling /data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLEDescriptorMap.cpp.o
Compiling /data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLEDevice.cpp.o
Compiling /data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLEHIDDevice.cpp.o
Compiling /data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLERemoteCharacteristic.cpp.o
In file included from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLECharacteristic.h:19,
                 from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLEDescriptor.h:14,
                 from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLE2904.h:13,
                 from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLE2904.cpp:15:
/data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
In file included from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLECharacteristic.h:19,
                 from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLEDescriptor.h:14,
                 from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLE2902.h:13,
                 from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLE2902.cpp:15:
/data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
*** [/data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLE2904.cpp.o] Error 1
*** [/data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLE2902.cpp.o] Error 1
In file included from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLEAdvertising.h:15,
                 from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLEAdvertising.cpp:21:
/data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
In file included from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLERemoteDescriptor.h:18,
                 from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLERemoteCharacteristic.h:18,
                 from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLERemoteService.h:16,
                 from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLEClient.h:19,
                 from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLEClient.cpp:13:
/data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
In file included from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLEDescriptor.h:16,
[... 67 lines cut ...]
/data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLERemoteCharacteristic.cpp:409:34: error: exception handling disabled, use -fexceptions to enable
   throw BLEDisconnectedException();
                                  ^
*** [/data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLERemoteCharacteristic.cpp.o] Error 1
/data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLEDevice.cpp: In static member function 'static void BLEDevice::whiteListAdd(BLEAddress)':
/data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLEDevice.cpp:490:75: error: too few arguments to function 'esp_err_t esp_ble_gap_update_whitelist(bool, uint8_t*, esp_ble_wl_addr_type_t)'
  esp_err_t errRc = esp_ble_gap_update_whitelist(true, *address.getNative());  // True to add an entry.
                                                                           ^
In file included from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLEDevice.cpp:17:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h:1478:11: note: declared here
 esp_err_t esp_ble_gap_update_whitelist(bool add_remove, esp_bd_addr_t remote_bda, esp_ble_wl_addr_type_t wl_addr_type);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLEDevice.cpp: In static member function 'static void BLEDevice::whiteListRemove(BLEAddress)':
/data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLEDevice.cpp:504:76: error: too few arguments to function 'esp_err_t esp_ble_gap_update_whitelist(bool, uint8_t*, esp_ble_wl_addr_type_t)'
  esp_err_t errRc = esp_ble_gap_update_whitelist(false, *address.getNative());  // False to remove an entry.
                                                                            ^
In file included from /data/cinema/.piolibdeps/cinema/ESP32 BLE Arduino/src/BLEDevice.cpp:17:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h:1478:11: note: declared here
 esp_err_t esp_ble_gap_update_whitelist(bool add_remove, esp_bd_addr_t remote_bda, esp_ble_wl_addr_type_t wl_addr_type);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [/data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLEClient.cpp.o] Error 1
*** [/data/cinema/.pioenvs/cinema/lib77d/ESP32 BLE Arduino/BLEDevice.cpp.o] Error 1
========================= [FAILED] Took 16.38 seconds =========================

```

gitolicious, 2022-12-17:
Same here. A minimal config YAML to reproduce the issue:
```yaml
esphome:
  name: ble-keyboard

esp32:
  board: esp32dev

external_components:
  - source: github://dmamontov/esphome-blekeyboard

ble_keyboard:
```
Going back to ESPHome 2022.11.5, it is working fine. I suspect the change on _core Arduino and ESP-IDF versions used for the ESP32 based devices_ (https://esphome.io/changelog/2022.12.0.html) is to blame.

dmamontov, 2022-12-19:
@LucaLa2001 @gitolicious Fixed in version: https://github.com/dmamontov/esphhome-blekeyboard/releases/tag/v2.3.1

closed 2022-12-19 by dmamontov

gitolicious, 2022-12-19:
Confirmed. Thanks for the quick fix!