dmamontov/esphome-blekeyboard #18: Press and hold down key

gitolicious, 2022-12-17:
**Describe the bug**
How can I press and hold a key? For example I would like to hold down the space bar for five seconds. Would have expected this to work:

```yaml
button:
  - platform: template
    name: "Hold Space"
    id: hold_space
    on_press:
      then:
      - ble_keyboard.press:
          id: myblekeyboard
          code: 0x20 # space
      - delay: 5000ms
      - ble_keyboard.release: myblekeyboard
```
It results in just the space bar being sent and immediately relesed. Did I misunderstand the `press` function in that it doesn't hold down the key until `release` is called? When would you use the `release` function otherwise?
 
**Please complete the following information:**
 - ESPHome version: 2022.11.5
 - Component version: latest? _(how to find out? I simply put `source: github://dmamontov/esphome-blekeyboard`)_

dmamontov, 2022-12-19:
@gitolicious In your case, a combination of one button should help:

```
button:
  - platform: template
    name: "Hold Space"
    id: hold_space
    on_press:
      then:
      - ble_keyboard.combination:
         id: myblekeyboard
         delay: 5
         keys:
           - 0x20
```

gitolicious, 2022-12-19:
I guess for that exact use case it would actually work - at least I can see a lot of space key events when putting `1000` instead of `5`) I don't fully understand why this works with space (`0x20`) but not with `"o"` where two individual letters are being printed.
Unfortunately there are two extensions I would need for my actual use case:
- I need to hold down Ctrl + Space (push to talk in MS Teams)
- I don't know how long to hold down before releasing the keys (there is a physical button as input)
Any ideas how to solve that?