dmamontov/esphome-blekeyboard #36: [BUG] Compilation Failure with esphome 2024.9.0

con247, 2024-10-15:
Hello,

I am having an issue compiling (new installation, have not previously used).

Here is my YAML (this is acting as an IR-> home assistant device, but I would like to add BLE keyboard functionality to hopefully add some IR control to PS5.

`substitutions:
  name: ESP_NAME
  friendly_name: Media Room IR Receiver

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.9.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

external_components:
  - source: github://dmamontov/esphome-blekeyboard

esp32:
  board: esp32dev
  framework:
    type: arduino

remote_receiver:
  pin:
    number: GPIO23
    inverted: true
  dump: all

binary_sensor:
  - platform: remote_receiver
    name: "IR Input - Sharp VOL+"
    pronto:
      data: "0000 006D 0010 0000 000B 0044 000B 001E 000B 001E 000B 001E 000B 001E 000B 0044 000B 0044 000B 001E 000B 0044 000B 001E 000B 0044 000B 0044 000B 0044 000B 001E 000B 0044 000B 0181"  
    filters:
      - delayed_off: 200ms
  - platform: remote_receiver
    name: "IR Input - Sharp VOL-"
    pronto:
      data: "0000 006D 0010 0000 000B 0044 000B 001E 000B 001E 000B 001E 000B 001D 000B 001E 000B 0044 000B 001E 000B 0044 000B 001E 000B 0044 000B 0044 000B 0044 000B 001D 000B 0044 000B 0181"
    filters:
      - delayed_off: 200ms
  - platform: remote_receiver
    name: "IR Input - Sharp Chan+"
    pronto:
      data: "0000 006D 0010 0000 000C 0044 000B 001D 000B 001D 000B 001D 000B 001D 000B 001E 000B 0044 000B 0044 000B 0044 000B 001E 000B 0044 000B 0044 000B 0044 000B 001D 000B 0044 000B 0181"
    filters:
      - delayed_off: 200ms
  - platform: remote_receiver
    name: "IR Input - Sharp Chan-"
    pronto:
      data: "0000 006D 0010 0000 000C 0044 000B 001D 000B 001D 000B 001D 000B 001D 000B 0044 000B 001D 000B 0044 000B 0044 000B 001E 000B 0044 000B 0044 000B 0044 000B 001D 000B 0044 000B 0181"
    filters:
      - delayed_off: 200ms
  - platform: remote_receiver
    name: "IR Input - Sharp PWR"
    pronto:
      data: "0000 006D 0010 0000 000C 0044 000B 001D 000B 001D 000B 001D 000B 001D 000B 0044 000B 001E 000B 001D 000B 0044 000B 001E 000B 0044 000B 0044 000B 0044 000B 001E 000B 0044 000B 0181"
    filters:
      - delayed_off: 1000ms
  - platform: remote_receiver
    name: "IR Input - Sharp MUTE"
    pronto:
      data: "0000 006D 0010 0000 000B 0044 000B 001E 000B 001D 000B 001D 000B 001D 000B 001E 000B 001D 000B 001D 000B 0044 000B 001E 000B 0044 000B 0044 000B 0044 000B 001D 000B 0044 000B 0181"
    filters:
      - delayed_off: 1000ms
# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

ble_keyboard:
  id: my_ble_keyboard
  name: "Logitech"
  manufacturer_id: "Logitech"
  battery_level: 100
  reconnect: true
  buttons: true
  use_default_libs: true

button:
  - platform: template
    name: "Enter"
    id: key_enter
    icon: "mdi:keyboard-return"
    on_press:
      then:
        - ble_keyboard.press:
            id: my_ble_keyboard
            code: 0xB0
        - ble_keyboard.release: my_ble_keyboard`

Here is my compilation error:

`INFO ESPHome 2024.9.2
INFO Reading configuration /config/esphome-web-c66cb0.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-c66cb0 (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.4
|-- ESP32 BLE Arduino @ 1.0.1
|-- NimBLE-Arduino @ 1.4.0
|-- ESP32 BLE Keyboard @ 0.3.2
Compiling .pioenvs/esphome-web-c66cb0/lib37c/ESP32 BLE Arduino/BLE2902.cpp.o
Compiling .pioenvs/esphome-web-c66cb0/lib37c/ESP32 BLE Arduino/BLE2904.cpp.o
In file included from .piolibdeps/esphome-web-c66cb0/ESP32 BLE Arduino/src/BLECharacteristic.h:19,
                 from .piolibdeps/esphome-web-c66cb0/ESP32 BLE Arduino/src/BLEDescriptor.h:14,
                 from .piolibdeps/esphome-web-c66cb0/ESP32 BLE Arduino/src/BLE2904.h:13,
                 from .piolibdeps/esphome-web-c66cb0/ESP32 BLE Arduino/src/BLE2904.cpp:15:
.piolibdeps/esphome-web-c66cb0/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
In file included from .piolibdeps/esphome-web-c66cb0/ESP32 BLE Arduino/src/BLECharacteristic.h:19,
                 from .piolibdeps/esphome-web-c66cb0/ESP32 BLE Arduino/src/BLEDescriptor.h:14,
                 from .piolibdeps/esphome-web-c66cb0/ESP32 BLE Arduino/src/BLE2902.h:13,
                 from .piolibdeps/esphome-web-c66cb0/ESP32 BLE Arduino/src/BLE2902.cpp:15:
.piolibdeps/esphome-web-c66cb0/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
*** [.pioenvs/esphome-web-c66cb0/lib37c/ESP32 BLE Arduino/BLE2904.cpp.o] Error 1
*** [.pioenvs/esphome-web-c66cb0/lib37c/ESP32 BLE Arduino/BLE2902.cpp.o] Error 1
========================== [FAILED] Took 3.98 seconds ==========================`

Thanks in advance for any assistance.

con247, 2024-10-17:
Compilation is failing still on the newly released 2024.10.0

skupu, 2024-10-18:
So for me compilation works, but the board doesnt boot